DFRobot/uPyCraft #1: Please post the source code.

nissimzur, 2017-10-12:
Please post the source code.
What the point of apps if no one can contribute to it?

ouki-wang, 2017-10-17:
@nissimzur thanks for your suggestion
We will post the source code after app stabilization, possibly in December.

Avi-TelnT, 2017-10-17:
The Idea is that we can help in apps debug and improve it too. 
So, please update it now, it still a good point to start and no need to wait to December.
I can send you many examples of esp32 code, you can add too. 
Thank you
nissim.zur@gmail.com

elf55, 2017-11-27:
GITHub is not meant to be a repository for closed source executables.

vanminh0910, 2017-11-29:
uPyCraft is a good tool and it's good to publish the source code so that everyone can contribute. By doing that, you gain the trust from community and make it more popular in uPython community.

nbxtruong, 2017-12-23:
This is the end of December. So, When do you open the source code?

nbxtruong, 2017-12-23:
@Avi-TelnT Do you have example for AWS IoT? That mean, How to Public and Subscribe a Topic by using MQTT? and how to update/get Thing Shadows?

Thank you so much ;)

Avi-TelnT, 2017-12-23:
I do not have....
This is demo from some loboris mP. I never tested it.

def conncb(task):
    print("[{}] Connected".format(task))

def disconncb(task):
    print("[{}] Disconnected".format(task))

def subscb(task):
    print("[{}] Subscribed".format(task))

def pubcb(pub):
    print("[{}] Published: {}".format(pub[0], pub[1]))

def datacb(msg):
    print("[{}] Data arrived from topic: {}, Message:\n".format(msg[0], msg[1]), msg[2])
mqtt = network.mqtt("loboris", "loboris.eu", user="wifimcu", password="wifimculobo", cleansession=True, connected_cb=conncb, disconnected_cb=disconncb, subscribed_cb=subscb, published_cb=pubcb, data_cb=datacb)

mqtts = network.mqtt("eclipse", "iot.eclipse.org", secure=True, cleansession=True, connected_cb=conncb, disconnected_cb=disconncb, subscribed_cb=subscb, published_cb=pubcb, data_cb=datacb)

NanaWangDFR, 2018-01-04:
@nbxtruong   Well, the deadline is 12th, December. We are still regulating the codes. Thanks for your note.

nbxtruong, 2018-01-06:
Thank you @NanaWangDFR, I am waiting for it. I hope it can run fine on ubuntu.

ouki-wang, 2018-01-12:
@nbxtruong @Avi-TelnT @vanminh0910 @nissimzur come to check the source code!
https://github.com/DFRobot/uPyCraft_src

mrgithub, 2018-01-17:
You should put the source in /uPyCraft it will make it easier for everyone to find it

Avi-TelnT, 2018-01-17:
@ouki-wang
Thank you for your great work.
You describe how to run it? or convert to exe?

ouki-wang, 2018-01-17:
@Avi-TelnT @mrgithub We are writing readme.md, which will describe the way to run with python Idel and pack src to .exe in detail.

nbxtruong, 2018-04-25:
Thank you @ouki-wang I am going to check it.

closed 2018-11-02 by ouki-wang