DFRobot/uPyCraft #2: How do update to latest firmware/bin file

yomasa, 2017-11-21:
I tryed loading the newest bin file for esp32  from micropython.org and uPyCraft keeps wanting to redo it?

ouki-wang, 2017-11-24:
The latest firmware of UpyCraft is published in 15th, Nov. You can check it by os.uname().
If you use the firmware of micropython.org, you also need to chck the information of os.uname().
If the micropython.org firmware date is earlier than UpyCraft firmware date, you should choose 'No Check' in checkfireware of tools->preferense->config.
uPycraft firmware include: bluetooth ， iis， sd，  necir

closed 2018-11-02 by ouki-wang

khueyeuqaqc, 2018-11-12:
Impressive work @ouki-wang 
Your build seems to be the only BLE-enabled I can found 
I am wonder if the implementation of BLE occupy 40K more RAM usage . 
In the original build , we get 92K of free mem . 
It is now only ~ 44K .
BTW , have you got any examples for BLE application .
Thanks .