DFRobot/uPyCraft #9: uPyCraft.exe not working windows 10

MichaelGGH, 2018-05-01:
Hello,

when i try to run the uPyCraft.exe on Windows 10 I get error:

code can't be executed because MSVCR100.dll not found.

What am I doing wrong? I trieed multiple download links from the internet.

[image]

ouki-wang, 2018-05-02:
We have already tested uPyCraft in 64bit and 32bit win10, the result is alright. Could you please figure out the PC type? Well, you can also try downloading the new version: uPyCraft_V0.30.exe.

jhlucky, 2018-05-02:
You can download this component by yourself.

MichaelGGH, 2018-05-02:
Thank you very much for replying,

I forgot to mention that im on Windows 10 Pro.

Not sure what a machine Type is, this is my build:
Windows 10 Pro Version 1709
CPU: Ryzen 5 1600x
Motherboard: MSI B350 Gamming Pro Carbon
RAM: 16 GB  
One HDD and one SSD
Nvidia 1060 GPU

I downloaded the hole repository. But uPyCraft.exe and uPyCraft_V0.30.exe both bring the error.

[image]

jhlucky, 2018-05-03:
@MichaelGGH In the package of uPyCraft, it haven't packed dll files. Which means your computer lacks of this library, you should download the library first then execute uPyCraft. If you still cannot open uPyCraft when the dll files has already be downloaded, I will help you to test the system as yours then try again.

MichaelGGH, 2018-05-03:
Thank you very much for your relply. It solved the issue. 

I had to install: 

Microsoft Visual C++ 2010 Redistributable Package (x86) 
Microsoft Visual C++ 2010 Redistributable Package (x64) 

Now I can start the uPyCraft.exe

Turned out that the file was not missing from the download but from my system just like you said.
Googel missing MSVCR100.dll could have solved the problem. But I did not now what a .dll is and
asumed that is a file from the uPycraft project.

sorry for me beeing a noob and thanks to everyone.

closed 2018-05-03 by MichaelGGH

Jibun-no-Kage, 2022-08-04:
On Windows 11 Pro, even with install of Microsoft Visual C++ 2010 SP1, no joy.  How about authors of uPyCraft help out here?

peterk62, 2023-04-20:
Windows 11 home - 64 bit OS. I installed the 64 bit redist package, but no joy. Installing the 32 bit redist package made it work.

Hdekkers-hwd, 2024-05-16:
Windows 11 pro , I tried to install upycraft v1.0 and 1.1. First the install told me that file MSVCR100.dll was missing. After installing Microsoft Visual C++ 2010 Redistributable Package (x64) i revceive the message: "toepassingsfout 0xc000007b on both versions. Any idea what is missing?