DFRobot/uPyCraft #11: Truncated text on fast response at ESP32

jkDesignDE, 2018-05-07:
Using version 0.30 on ESP32 with MicroPython 1.9.1.

The REPL works fine ... but if a response is send from the ESP32 with high speed like in a trackback or a tab-completion the text is truncated and/or crippled.

Example (tab completion on sys):
```
>>> import sys
>>> sys.
__name_   stdin   5stderr          modules         print_exception
>>> sys.
__name_          stdout
stderr          modules         print_exception
>>> sys.
__nam          stdout
stderr          modules         print_exception
>>> sys.
```
Note: Using a terminal program (Putty) the response is correct.
Note: On ESP8266 all works fine.

jhlucky, 2018-05-07:
Multithreading has already be adopted. Well, to enable IDE supports multiple functions, it maybe lead to resource conflict and package lost in reading serial port. We are fixing this problem, and it will be solved in the next version. If it is unconvenient for you to use 0.30 uPyCraft, we'd love to recommend 0.29 uPyCraft, which is more stable than the new one.

jkDesignDE, 2018-05-07:
Response from version 0.29:

```
>>> import sys
>>> sys.
__name__$
recv unexpected word.
recv unexpected word.
recv unexpected word.
recv unexpected word.
recv unexpected word.
recv unexpected word.5stderr          modules         print_exception
```

closed 2018-11-02 by ouki-wang