DFRobot/uPyCraft #16: Not work in Ubuntu 18.04

mcdum, 2018-07-20:
Hi,
I try to launch uPyCraft_linux_V0.30 and uPyCraft_linux_V1.0 but none work.
The error is:
```
mcd@mcd-desktop:~/Programe/uPyCraft$ ./uPyCraft_linux_V1.0
Traceback (most recent call last):
File "uPyCraft.py", line 2, in <module>
File "/usr/local/lib/python3.5/dist-packages/PyInstaller/loader/pyimod03_importers.py", line 714, in load_module
ImportError: /tmp/_MEI13fSvN/libz.so.1: version `ZLIB_1.2.9' not found (required by /usr/lib/x86_64-linux-gnu/libpng16.so.16)
[3992] Failed to execute script uPyCraft
```
The library is installed.
Thanks in advance.
Mihai D.

ouki-wang, 2018-07-23:
@mcdum The uPyCraft works fine in Ubuntu 16.04. Other versions have not been tested and require users to improve themselves.

palto42, 2018-12-31:
I found a modified version for Ubuntu 18.04 with QT5 https://github.com/jiapei100/uPyCraft_PyQt5

tensau, 2020-04-04:
OK so it is a totally useless tool if you can't use it with current linux versions.