DFRobot/uPyCraft #17: Icon File doesn't work on macOS

PlasmaSoftUK, 2018-07-29:
Minor guy issue but as per title the .ico in Resources folder doesn't work on MacOS needs to be a .icns file.

PlasmaSoftUK, 2018-07-30:
Ok, let me see if I can explain this a bit better with a picture. 

[image]

On macOS the 'uPyCraft' App Icon doesn't show up. It has the pain default no icon picture. But if you look inside "uPyCraft/Contents/Resources/" there is a "logo.ico" which is the correct place for the logo on macOS however it needs to be an IconSet called "logo.icns" and correctly referenced from the "info.plist" in Contents.

jhlucky, 2018-07-31:
Thank you for your replying, I will debug it after two days. You are right. The next version will be uplaoded after debugging.

todocono, 2019-04-03:
This bugs still exists

OptimusGREEN, 2020-05-04:
and still exists...