DFRobot/uPyCraft #19: NodeMCU-32S need burn firmware if the connection is disconnected and reconnected.

alaj, 2018-10-11:
Board: NodeMCU-32S
uPyCraft version: 0.30, 1.1
Situation: connect the board with computer, and use uPyCraft to burn firmware(esp32). Then I can download and run program. But if I click disconnect button on uPyCraft and then click connect button again, it pop the dialog and ask me to burn firmware again. Every time I reconnect, I need to reburn firmware again. Is my operation wrong?

jhlucky, 2018-10-12:
Upy comes with the latest firmware verision of ESP8266, 0721(note: you can check it throgh os.uname()). It will prompts upgrade if your version of the firmware of esp8266 lower than 0721. If you don't want to upgrade your firmware, you can set to ignore fireware upgrading at  Tools->Preferences->config.

White-Goblin, 2018-11-25:
I have the same issue. It is able to connect the first time without problems, but after disconnecting, I get prompted to burn firmware again. Changing the preferences does not fix the issue.

alaj, 2018-12-12:
I set upy to ignore upgrading firmware. Still not working.

SilvesterHsu, 2019-02-02:
I find that the connection may have `time_out` strategy. The connection time starts counting from the boot of my esp. If the time spending on the main programme is longer than a specific time, it would fail during connecting.

smith-scot, 2019-02-11:
I have a similar issue; however I only notice the problem after I have connected the wifi.  I can connect serially several times with no problems noticed.  It is only after I have an active WiFi connection that I have the problem, and even after uPyCraft requests to burn firmware, I can go to another serial program (MobaXterm) and connect to the ESP32.  There doesn't seem to be anything wrong with the ESP32.  Any thoughts?

SilvesterHsu, 2019-02-12:
@smith-scot I guess this will be related to the ESP runtime. If you use the wifi connection example provided by [Micropython](https://docs.micropython.org/en/latest/esp8266/tutorial/network_basics.html), you will find a process waiting for a wifi connection.
```python
while not sta_if.isconnected():
    pass
```
This will wait for the router DHCP to assign an IP address, however this may exceed the IDE's timeout. 

I use a MacBook, but the experience of this software on the mac is not very good, you can try the Windows version. And the version on Github seems to be slower to update, they have the same project on [gitee](https://gitee.com/dfrobot/upycraft/), and the [v1.1 version](http://download3.dfrobot.com.cn/uPyCraft/uPyCraft_V1.1.exe) is released.

smith-scot, 2019-02-12:
@SilvesterHsu I have been doing some additional testing, and the problem seems to be corrected when I skip the connect() function.  I will need to perform some test so see if I can prevent this, but like you stated, the problem seems to be associated with a failure to sign into the Wifi network.  I am currently using v1.1, I don't an option to increase the IDE's timeout.  Does this option exist?

SilvesterHsu, 2019-02-16:
@smith-scot Sorry, I didn't find that the timeout is optional. It is estimated that this is related to how the software is connected.
In addition, I also use [ESPlorer](https://github.com/4refr0nt/ESPlorer) frequently, which seems to be software developed by the Russians. ESPlorer does not have a timeout problem, it can load the program quite smoothly, and also provides the RTS button. It was always a pleasant time to develop on ESPlorer. However, the only problem is the file uploading bug.
The original author of ESPlorer seems to have stopped updating the software. Uploading the file sometimes fails, but Josverl modified the upload function and [his version](https://github.com/Josverl/ESPlorer/tree/MT-Build) will be much more stable.

chrissyast, 2019-04-13:
I am finding this same issue on uPyCraft v1.1, I changed CheckFirmware to `no check` but still get the prompt for firmware every time. Just wondering if any of you found any solutions since the last post? I don't see where is the `connect()` function that @smith-scot mentions.

I guess I will try ESPlorer and see how it works.

larslundh, 2019-06-05:
> I have a similar issue; however I only notice the problem after I have connected the wifi. I can connect serially several times with no problems noticed. It is only after I have an active WiFi [...]
_________________________________________________________
Hello, I just want to Second that I have identical issue.
Just like you said, Serical communication works fine (as well as Webrepl if the micro is loading the wifi code from micropython.org) to add: 
my suggestion is to delete the wifi code from your micro via Serial or Webrepl
control + C to stop it doing stuff, if its locked then...
>>>import os
>>>os.remove(main.py)
GG try re-connect (hopefully you put the wifi code there...)

Aschhoff, 2020-05-03:
On Windows use my Tool:  
https://github.com/Aschhoff/ESP-Autostart-Changer