DFRobot/uPyCraft #30: Does not run on Linux, libz.so.1: version ZLIB_1.2.9 not found

probonopd, 2019-03-03:
On `xubuntu-18.04-desktop-amd64.iso`, getting

```
me@host:~$ '/home/me/Downloads/uPyCraft_linux_V1.0'
Traceback (most recent call last):
  File "uPyCraft.py", line 2, in <module>
  File "/usr/local/lib/python3.5/dist-packages/PyInstaller/loader/pyimod03_importers.py", line 714, in load_module
ImportError: /tmp/_MEIrooXex/libz.so.1: version `ZLIB_1.2.9' not found (required by /usr/lib/x86_64-linux-gnu/libpng16.so.16)
[8864] Failed to execute script uPyCraft
me@host:~$ '/home/me/Downloads/uPyCraft_linux_V0.30'
Traceback (most recent call last):
  File "uPyCraft.py", line 2, in <module>
  File "/usr/local/lib/python3.5/dist-packages/PyInstaller/loader/pyimod03_importers.py", line 714, in load_module
ImportError: /tmp/_MEIk86I33/libz.so.1: version `ZLIB_1.2.9' not found (required by /usr/lib/x86_64-linux-gnu/libpng16.so.16)
[8913] Failed to execute script uPyCraft

```

crisconru, 2019-04-30:
I've got the same problem in Elementary OS (Ubuntu 18.04)

leozlee, 2019-07-11:
Does anyone how to deal with it?

probonopd, 2019-07-13:
Build on trusty or xenial, but not on a newer OS version.

bepstein111, 2019-09-25:
Same problem on Xubuntu 19.04...sighhhh

mf01, 2020-04-19:
Same on Ubuntu 19.10

mohitchaniyal, 2021-12-07:
anyone fixed the issue