DFRobot/uPyCraft #31: uPyCraft v1.1 Frequent issue after download completed, NameError: name 'os' isn't defined

dachshund-digital, 2019-03-05:
Did not get this issue with older version of uPyCraft, but frequently see it on version 1.1 on Windows 10 using Python 3.7.2.

Ready to download this file,please wait!
.....................
download ok

os.listdir('.')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'os' isn't defined
>>> 
reflush tree false

There is definitely something else going on with the IDE, when I compare files in my workspace compared to device directory, using ESP8266 devices (WeMos and LoLin variants), I notice that the IDE is NOT synchronizing the files consistently even when no errors reported.  I have to disconnect and reconnect the device, then download files again, then use the device reset button to ACTUALLY see the correct variant of the code run.

Never had this issue with these same devices using Arduino IDE, so seriously doubt the multiple devices I have been using are at issue here.

This one is really driving me nuts.  Any help appreciated.

sjgcornish, 2020-02-09:
Did you every find out what this was and how to fix it. I have the same problem, although the programs upload and run OK.

Arvind595, 2020-03-25:
It means the new file was uploaded to your board successfully. You can notice that it printed the “download ok” message. Press the ESP on-board “EN/RST” button to restart your board and re-run the new uploaded script from the beginning.
check out more here https://randomnerdtutorials.com/getting-started-micropython-esp32-esp8266/

uteyechea, 2020-05-13:
I had a similar problem. For some unknown reason, if I place the problematic code inside a function, then execute the function and go figure it worked like charm.

a1h2med, 2020-06-29:
I came through this problem earlier today, And honestly IDK why this happened, but when I uninstalled (UPycraft) and deleted all of its' files then restarted my laptop, opening a serial then close it... After multiple trials suddenly it was solved (I don't know how).
SO It's not a technical solution it's just something like (trial and error), I searched everywhere btw and unfortunately, I couldn't find any solution.

drberthol, 2021-02-09:
I close uPycraft, disconnect the esp32 and it's resolve the problem. Just in case it's useful for someone