DFRobot/uPyCraft #35: 软件还有人维护吗？？？！！！ Please fix serial port debugging

CHN-STUDENT, 2019-06-23:
先把串口调试输出做好吧！！！
调试信息都显示不出来!!!
The serial port debugging output can not display all info correctly! It is a big issue for our user to debug!! Please fix it! Thanks!
See here:https://github.com/DFRobot/uPyCraft/issues/26

ouki-wang, 2019-06-24:
Currently unmanned

barbudor, 2019-06-24:
@CHN-STUDENT
What is your problem? 
Your report is useless. A bug report must include all details to allow reproduction. 
Please details
- target board
- pc/Mac operating system
- what are you doing? 
- what happens? 
- how this is différent from expectation?
- have you seek help on the forum before raising an issue?

CHN-STUDENT, 2019-06-24:
@barbudor Ok, thanks , i update it.

barbudor, 2019-06-24:
Hi @CHN-STUDENT 
First let me clarify that I am not involved with DFRobot or with the development of this software.
I am just a user, and while we all know that there are a few issues, the author made it clear that he/she doesn't have the resource for fixing this small bugs.
So either (1) we live with it as it is, (2) we walk away and look for a better software, or (3) we fork and fix it ourselves. As I'm not yet able to do (3), I've chosen (1).

However most of the problems that you are raising are either known or comes from a lack of experience with either uPyCraft or MicroPython.
There are workaround for some.

1) This is know and the workaround has already been given in the other issues. You have to install the font as Administrator, not as user.

2) You still doesn't say
- Which ESP32 board you are using (although it shouldn't make much difference)
- Which version of Micropython you have flashed
- If there is a code running on the board
I generally have no problems to connect to my boards.
For uPyCraft to work, it must have access to the REPL. On startup, uPyCraft is sending a `Ctrl-C` to the board to stop any running program and get access to the REPL. If this doesn't work and uPyCraft doesn't receive the REPL `>>>`  prompt, it believes that the board if brand new and that MicroPython needs to be flashed.
Why isn't your board responding ? As you are not providing enough details I can only guess from my experience:
- MicroPython it is crashed
- your Python software has disabled Keyboard Interrupt.
- your board is crashed
uPyCraft can't do anything more than sending `Ctrl-C`. If the board doesn't behave as expected, uPyCraft can't do anything about it.
You may try to reset the board and retry to connect until uPyCraft successfully get access to the REPL.
It may not be possible and the board may requires reflashing the board to erase your program (if your program is the fault).
As you don't provide a clear procedure how to reproduce your problem, I can't give specific advice.

3) I never faced this problem and you don't provide a clear procedure to reproduce it. So I can't help.
But it looks like uPyCraft didn't get response as expected from your board and consider that there is a program running that doesn't stop.

4) Obviously your code `mqtt.py` is crashing the board. Is it because of the code or is it because your board is faulty, I don't know.
But once again this is not the fault of uPyCraft.
Note that I never use "download and run". I always "download" and then use "import" to run the code.
I don't like much the way `run` is done. No rational, just a personal feeling.

other 4) It is very likely that your `boot.py` doesn't include `import os.py`
On 1st start/connection, uPyCraft send `import os`to the board so the `os` module is available. But if you reset the board by Ctrl-D then this import has been cleared unless you have it in `boot.py`.
Of course, we could ask to the author to always send `import os` before any other command that needs it. But that will impact performance to send this command when not needed.
So add `import os` in your boot.pi and you will be good.

5) `Ctrl-S` NEVER save the file to the device. `Ctrl-S` always save the file to your computer IF the file has been opened from the computer (workspace).
If you open the board from the device, you can't do anything with it unless you "save as" an new file on the computer.
Only `Download` can write a file to the board.

Let me know if you have further questions.

Cheers.