DFRobot/uPyCraft #36: SPI device already in use

FlyingThunder, 2019-06-26:
Windows 8
LOLIN32
V1.0

When flashing my code after already having flashed it once, it will always tell me
`E (438644) spi_master: spi_bus_initialize(108): host already in use[0m
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<string>", line 25, in <module>
OSError: SPI device already in use`

This (seems?) to have been fixed here:
https://github.com/micropython/micropython/commit/da72bb6833ce8e43bc663e3d48596b9209d05464

I just want to ask if there is a solution for upycraft specifically, as i dont know what micropython version is used and if/how i can make upycraft use the newer micropython version

barbudor, 2019-06-26:
Hi @FlyingThunder
Although you don't tell, I believe that you are using 'download and run' 
The problem doesn't lies into uPyCraft how micropython is working and your script

With 'download and run' you are executing your script twice just like if you were typing it line by line in the REPL. 
If your code doesn't explicitly releases the resources allocated during 1st execution, then the 2nd execution fails to allocate the same resource again. This is what happens with the SPI bus here. 

There are 2 solutions
1) improve your code to release properly all resources before ending so it can be available for next run
2) reboot the board with ctrl-d before doing 'download and run' 

The 2nd solution is undoubtedly the simplest. 

Hope this helps

FlyingThunder, 2019-06-26:
I tried machine.reset() and power cycling it, assuming that this would hard reset the device and clear the executed code. The error still appeared afterwards, i will try ctrl-d tomorrow.

barbudor, 2019-06-26:
Ok, so may be I misunderstood what happens. Ctrl-D and machine.reset() should do the same and reboot the MPy machine so all resources should be freed. If you reboot or power cycle the board between the 2 'download and run', I don't understand why the resources should still be in-use.
If you still have the problem, please details all minimum steps that allow to recreate the problem.

FlyingThunder, 2019-08-06:
Hello, sorry for the late reply - i press Ctrl+D, and this shows up: 
`>>> 
PYB: soft reboot
[0;31mE (243801) spi_master: spi_bus_initialize(108): host already in use[0m
Traceback (most recent call last):
  File "boot.py", line 22, in <module>
OSError: SPI device already in use
OSError: [Errno 2] ENOENT
MicroPython v1.9.1-224-g83d3f3f-dirty on 2017-12-12; ESP32 module with ESP32
Type "help()" for more information.
>>> `

This is the code im running: 
boot.py: https://pastebin.com/du8gmSMX
driver.py (called by boot): https://pastebin.com/3FSRmi99

barbudor, 2019-08-08:
Hi @FlyingThunder,
Sorry I'm on leave now with no devices with me to test ;) 

I feel you have to much in boot.py. Not sure this is the cause at all but in my case, I only have a minimal boot.py and the app would be in main.py

If you have an empty boot.py and no main.py and start copy line-by-line in the repl, do you get the same behavior?