DFRobot/uPyCraft #43: MacOS Ventura 13.1 exception on install/launch

frankcohen, 2023-01-16:
Downloaded from https://randomnerdtutorials.com/uPyCraftMac, get a crash error when opening the app on MacOS 13.1, 
2.3 GHz Quad-Core Intel Core i7, Powerbook laptop.

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               uPyCraft [43199]
Path:                  /private/var/folders/*/uPyCraft.app/Contents/MacOS/uPyCraft
Identifier:            uPyCraft
Version:               0.0.0 (???)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-01-16 09:34:36.5810 -0800
OS Version:            macOS 13.1 (22C65)
Report Version:        12
Bridge OS Version:     7.1 (20P2059)
Anonymous UUID:        A2D47672-A847-DE4A-4DFF-48BD3DDD62BE

Sleep/Wake UUID:       0FA0294B-5AAF-45B2-A0BB-579FF1A8461A

Time Awake Since Boot: 130000 seconds
Time Since Wake:       2481 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGSEGV)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	    0x7ff80a87822a __pthread_kill + 10
1   libsystem_pthread.dylib       	    0x7ff80a8aff7b pthread_kill + 263
2   libsystem_c.dylib             	    0x7ff80a7bc530 raise + 26
3   uPyCraft                      	       0x10e1199f7 0x10e115000 + 18935
4   uPyCraft                      	       0x10e117721 0x10e115000 + 10017
5   uPyCraft                      	       0x10e115ec4 0x10e115000 + 3780

Thread 1:
0   libsystem_pthread.dylib       	    0x7ff80a8abc58 start_wqthread + 0

Thread 2:
0   libsystem_pthread.dylib       	    0x7ff80a8abc58 start_wqthread + 0

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000000000000  rbx: 0x00007ff84e1e08c0  rcx: 0x00007ff7b1de9d78  rdx: 0x0000000000000000
  rdi: 0x0000000000000103  rsi: 0x000000000000000b  rbp: 0x00007ff7b1de9da0  rsp: 0x00007ff7b1de9d78
   r8: 0x0000000000000400   r9: 0x00000000000003ff  r10: 0x00007ff84e1e08c0  r11: 0x0000000000000246
  r12: 0x0000000000000103  r13: 0x00007ff7b1deac38  r14: 0x000000000000000b  r15: 0x0000000000000016
  rip: 0x00007ff80a87822a  rfl: 0x0000000000000246  cr2: 0x0000600001d48000
  
Logical CPU:     0
Error Code:      0x02000148 
Trap Number:     133

Binary Images:
    0x7ff80a870000 -     0x7ff80a8a9ff7 libsystem_kernel.dylib (*) <7c3dcc95-9f42-3c7c-8796-476ff67b9cf7> /usr/lib/system/libsystem_kernel.dylib
    0x7ff80a8aa000 -     0x7ff80a8b5ff7 libsystem_pthread.dylib (*) <5920e36f-53ec-33f0-b675-8ae48b58418c> /usr/lib/system/libsystem_pthread.dylib
    0x7ff80a778000 -     0x7ff80a800fff libsystem_c.dylib (*) <376f7cb7-6dd2-3e00-976f-77dd755bdb0d> /usr/lib/system/libsystem_c.dylib
       0x10e115000 -        0x10e11bfff uPyCraft (0.0.0) <32ab4b3f-56df-3c81-b3ee-2215ac410955> /private/var/folders/*/uPyCraft.app/Contents/MacOS/uPyCraft

External Modification Summary:
  Calls made by other processes targeting this process:
    task_for_pid: 0
    thread_create: 0
    thread_set_state: 0
  Calls made by this process:
[... 132 lines cut ...]
    "queue" : "com.apple.main-thread"
  }
},
  "trialInfo" : {
  "rollouts" : [
    {
      "rolloutId" : "5ffde50ce2aacd000d47a95f",
      "factorPackIds" : {

      },
      "deploymentId" : 240000238
    },
    {
      "rolloutId" : "63a0a6cfd1cdb67d083006d0",
      "factorPackIds" : {

      },
      "deploymentId" : 240000003
    }
  ],
  "experiments" : [

  ]
}
}

MKZwei, 2023-05-10:
I  have exactly the same issue with the version Mac OS 13.3.1. It's good to know that it might be a problem in the software and not the settings on my system.

faatemehch, 2023-08-26:
i have this problem too. did you fine any solution?

dtappia, 2023-12-20:
Still crashing with MacOS Sonoma 14.2. Here the full report:
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               uPyCraft [63750]
Path:                  /Applications/uPyCraft.app/Contents/MacOS/uPyCraft
Identifier:            uPyCraft
Version:               0.0.0 (???)
Code Type:             X86-64 (Native)
Parent Process:        Exited process [63748]
User ID:               501

Date/Time:             2023-12-20 14:16:53.8755 +0100
OS Version:            macOS 14.2 (23C64)
Report Version:        12
Bridge OS Version:     8.2 (21P2057)
Anonymous UUID:        73FFC734-1BA3-1DE5-C794-A37626A475C9

Sleep/Wake UUID:       699401C8-234D-462A-9F06-FA763E2E0EDD

Time Awake Since Boot: 44000 seconds
Time Since Wake:       1085 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x00000000676c6f6a
Exception Codes:       0x0000000000000001, 0x00000000676c6f6a

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [63750]

VM Region Info: 0x676c6f6a is not in any region.  Bytes before following region: 2786353302
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      10d80d000-10d814000    [   28K] r-x/rwx SM=COW  ...acOS/uPyCraft

Kernel Triage:
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   QtGui                         	       0x114b73b14 QAction::isEnabled() const + 4
1   QtGui                         	       0x114b5817c qt_mac_set_modal_state_helper_recursive(NSMenu*, NSMenu*, bool) + 412
2   QtGui                         	       0x114b58107 qt_mac_set_modal_state_helper_recursive(NSMenu*, NSMenu*, bool) + 295
3   QtGui                         	       0x114b5d10c QMenuBarPrivate::macUpdateMenuBarImmediatly() + 2492
4   QtGui                         	       0x114b269cf -[NSApplication(QApplicationIntegration) qt_sendPostedMessage:] + 287
5   QtGui                         	       0x114b26ac2 -[NSApplication(QApplicationIntegration) qt_filterEvent:] + 130
6   QtGui                         	       0x114b26b88 -[QNSApplication sendEvent:] + 56
7   AppKit                        	    0x7ff80f948b93 -[NSApplication _handleEvent:] + 65
8   AppKit                        	    0x7ff80f486d57 -[NSApplication run] + 640
9   QtGui                         	       0x114b3017a QEventDispatcherMac::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 2106
10  QtCore                        	       0x112c98eb8 QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 504
11  QtCore                        	       0x112c9c077 QCoreApplication::exec() + 199
12  PyQt4.QtGui.so                	       0x1139ad7b0 meth_QApplication_exec_(_object*, _object*) + 80
13  Python                        	       0x112019cd9 PyCFunction_Call + 233
14  Python                        	       0x1120a7e34 PyEval_EvalFrameEx + 38500
15  Python                        	       0x1120a88a0 _PyEval_EvalCodeWithName + 2400
16  Python                        	       0x1120a8a01 PyEval_EvalCode + 81
17  uPyCraft                      	       0x10d80f0c9 0x10d80d000 + 8393
18  uPyCraft                      	       0x10d80f69a 0x10d80d000 + 9882
19  uPyCraft                      	       0x10d80dec4 0x10d80d000 + 3780

[... 880 lines cut ...]
}

Model: MacBookPro16,1, BootROM 2020.61.1.0.0 (iBridge: 21.16.2057.0.0,0), 6 processors, 6-Core Intel Core i7, 2,6 GHz, 16 GB, SMC 
Graphics: Intel UHD Graphics 630, Intel UHD Graphics 630, Built-In
Graphics: AMD Radeon Pro 5300M, AMD Radeon Pro 5300M, PCIe, 4 GB
Display: Color LCD, 3072 x 1920 Retina, Main, MirrorOff, Online
Memory Module: BANK 0/ChannelA-DIMM0, 8 GB, DDR4, 2667 MHz, Micron, 8ATF1G64HZ-2G6E1
Memory Module: BANK 2/ChannelB-DIMM0, 8 GB, DDR4, 2667 MHz, Micron, 8ATF1G64HZ-2G6E1
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x7BF), wl0: Jul 10 2023 12:53:18 version 9.30.503.0.32.5.92 FWID 01-c06f991b
AirPort: 
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: USB 10/100/1000 LAN, Ethernet, en7
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: T2Bus
USB Device: composite_device
USB Device: Touch Bar Backlight
USB Device: Touch Bar Display
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Headset
USB Device: Ambient Light Sensor
USB Device: FaceTime HD Camera (Built-in)
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 63.5
Thunderbolt Bus: MacBook Pro, Apple Inc., 63.5

janelleenqi, 2024-05-12:
I have just tried to open my installed uPyCraft and I am facing the same issue (macOS Ventura 13 too). Have any workarounds been found?